bigtreetech/Manta-M8P #127: LEDs, Strip lights, Neopixels

KD-4-Lyfe, 2024-09-08:
Need to know if i need an external PSU or not.

What is the Max amps allowed by the manta M8P V2

was planning running  2 lots of about 30 RGBW WS2812B down either side of my printer and maybe  smaller progress bar.

bigtreetech, 2024-09-09:
Don't go over 2A since that rail is also powering the SBC.

closed 2024-09-09 by bigtreetech

KD-4-Lyfe, 2024-09-09:
Much appreciated, I'll run an external PSU, could I advise that be added the documentation also.

bigtreetech, 2024-09-09:
Good point. @BTTJUSTIN let's get this added to the manual. The specification is how much current the user can draw from the 5V rail.

oMTEo, 2024-09-09:
Dont Forget to look into this -> https://github.com/bigtreetech/Manta-M8P/issues/81

bigtreetech, 2024-09-09:
> Dont Forget to look into this -> #81

The provided response still stands. It is 100% accurate in spite of what the OP may think.