bigtreetech/Manta-M8P #137: How to use Manta M8P v2 with 12864 display

WayneManion, 2025-03-03:
I have a Sovol SV08. I want to convert it to use a Manta M8P v2 that I already have. I don't like touch screens. I don't want to run KlipperScreen. I want to use an old-school 12864 display and control wheel. 

What do I need to buy to be able to connect a 12864 display to this board? It doesn't have the normal EXP1+EXP2 connectors. As far as I can tell, the same pins are accessible through some kind of ribbon connector. How do I break this out to connect it to the normal EXP1+EXP2 connectors?

royalfifty8, 2025-03-05:
Yes you can use EXP connectors or I2C(only V2.0 for the old school's. You need to purchase either a CB1, CB2 or Raspberry Pi compute model or raspberry Pi 3,4, or 5. You will not being able to use HDMI or CSI interface otherwise. You can always remote from your computer as well. I do highly recommend grabbing yourself the CB1 at least. It's equivalent to the Raspberry Pi 3. Operating the board by itself tends to be a bit sluggish.  

P.S. You might run into some speed bumps when flashing the board, If it gives you trouble with the boot loader, check out Dr.Klipper, he has a great step by step guide.

WayneManion, 2025-03-05:
After more research that should have been needed, I think I found the most straightforward solution.

BTT Mini12864 v2.0 displays have the same 18-pin 1.0mm-pitch FPC connector as the M8P. So the most elegant solution is to get a BTT Mini12864 v2.0 display and an 18-pin  1.0mm-pitch FPC cable and connect them that way. 

I only have a couple Fysetc Mini 12864 v2.1 displays and several stock Ender 3 12864 displays on hand and none of them have the FPC connector. The product listing on Amazon for the BTT Mini12864 v2.0 doesn't show the FPC connector on the back (if you look closely at the Amazon listing's picture, it is screen printed "BIGTREETECH MINI12864 v1.0"). 

I believe all this to be true after squinting and counting pins on the M8P's FPC connector, measuring the width of the ribbon cable opening, and dividing the width by the pin count. This would have been a lot easier if BTT wrote better documentation. The docs for the M8P v2 and  the Mini12864 v2.0 both just call the connector an "FPC", leaving it to the user to count pins and measure pitches themselves.

royalfifty8, 2025-03-06:
I was going to mention I've tried messing with the Ender displays but ended up with the same issue and wasted time. And yes, I agree. Unfortunately, BTT's literature and CS don't possess the same grandeur as their boards do. Lol. Looking forward to what you come up!

WayneManion, 2025-03-11:
The BTT Mini12864 board does come with an 18-pin FPC 1mm-pitch cable that works with the Manta M8P. 

Sadly, the included cable was too short for my application. I can't fault BTT for that. This printer is huge and I have the mainboard mounted at the back of the printer. I bought some longer cables on Amazon and confirmed that 18-pin, 1-mm pitch is the secret sauce. I had to order a 400mm cable from AliExpress for my printer.  The only one I could find in the US was from Digi-Key and it was $12 plus shipping. 

Use this to get it working:
```
[display] # BTT Mini12864 v2.0 connected via FPC cable
lcd_type: uc1701
cs_pin: PG0
a0_pin: PF15
rst_pin: PF14
encoder_pins: ^PE10, ^PE15
click_pin: ^!PG1
contrast: 63
spi_software_miso_pin: PE13
spi_software_mosi_pin: PE14
spi_software_sclk_pin: PE12

[output_pin beeper]
pin: PE7

[neopixel mini12864] #	control Neopixel RGB in mini12864 display
pin: PF13
chain_count: 3
initial_RED: 0.1
initial_GREEN: 0.5
initial_BLUE: 0.0
color_order: RGB

##	Set RGB values on boot up for each Neopixel. 
##	Index 1 = display, Index 2 and 3 = Knob
[delayed_gcode setdisplayneopixel]
initial_duration: 1
gcode:
  SET_LED LED=mini12864 RED=0 GREEN=1 BLUE=1 INDEX=1 TRANSMIT=0
  SET_LED LED=mini12864 RED=0 GREEN=1 BLUE=1 INDEX=2 TRANSMIT=0
  SET_LED LED=mini12864 RED=0 GREEN=1 BLUE=1 INDEX=3
```