bigtreetech/Manta-M8P #138: No matter what I do, Klipper install hangs with CB2

LiamWabbit, 2025-03-06:
Klipper works as as far as typing sudo nand-sata-install but I get past the format stage and it just hangs on counting files. 
This is really winding me up now. I can't find any trouble shooting for this at all.
It klipper from BIQU itself. It should take 15 mins to install, I've wasted a day with it. Just trying to to find out how to fix it.
I've tried different SD cards. Even USB doesn't work with the manta. it read a port device, but the software from your site wont do any firmware stuff written in your instructions. Is there a fix for is, or should I send £150 worth of tech back! 
It is not meant to be this difficult! Install klipper to the emmc and then install the firmware for manta board using klipper and ssh
The time I've spent on this is costing me money

bigtreetech, 2025-03-06:
Just a heads up that this is not related to installing klipper. It sounds like this is OS related. Please provide info about exactly what OS image you are trying to use and, if possible, even a picture of where you see it hanging.

In terms of manuals, we don't produce video manuals for most products. We produce written form documentation.

royalfifty8, 2025-03-06:
@LiamWabbit This should help. Along with a few other tabs on his website. 

https://www-drklipper-de.translate.goog/doku.php?id=klipper_faq:flash_guide:stm32h743:btt_manta_m8n_usb&_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en-US