bigtreetech/Manta-M8P #141: Issues with 48 power supply

gilbertorconde, 2025-04-21:
I've a manta m8p v2 on a voron 2.4. If im using the same 24v power supply on all board and motors everything works fine. If i add a 48v to the A/B Motors whenever the toolhead starts moving i get a disconnection error. Sometimes from the toolhead, sometimes from the scanner (carthographer, btw). I have a CM4 emmc and TMC5160T PRO for the A/B motors all the rest is basically the stock from the formbot kit.
Btw, the toolhead starts moving, the disconnection always happens but only after some secconds. Some times it gets to home one of the axis.
Im wondering if any one here had the same issues and possible any help on debugging.
My printer works fine without the 48v, im just messing around to learn more about all of this.
My 48v power supply is this one:
https://aliexpress.com/item/1005005474634071.html?spm=a2g0o.order_list.order_list_main.154.2092caa4XRZQdO&gatewayAdapt=glo2bra

bigtreetech, 2025-04-21:
There are a few options for what could be causing this but they all lead back to EMI interference.

1. Your new PSU is electrically noisy and generating noise that is interfering with the communication signal to your toolboard. Check for parts of the wiring where the AB motor wires may run close to the comms wires.
2. The higher voltage being sent to the motors is creating larger electric fields around the AB wires which could also contribute towards EMI.

Are you using CAN or USB?

gilbertorconde, 2025-04-21:
I'm using usb.
Yeah I kinda suspected something like that. Do you know if add some sort of shield to these cables will work? Im using a drag chain from the gentry to the bottom of the print. Re routing the motor cables away from the USB one would be a big pain.
Many tks for your quick reply.

gilbertorconde, 2025-04-22:
its confirmed, going to close this issue. Many tks

closed 2025-04-22 by gilbertorconde